rvmn/SuperPleccer #7: AGPLv3 binary distribution without sources

derlaft, 2023-03-19:
Hi,

> Please do not clone and build the source code yet, the source is not up to date. See releases for prerelease versions, for now this is the only option to test/try

If I understand AGPL correctly, this is not allowed:

> Permissions of this strongest copyleft license are conditioned on making available complete source code of licensed works and modifications, which include larger works using a licensed work, under the same license. Copyright and license notices must be preserved. Contributors provide an express grant of patent rights. When a modified version is used to provide a service over a network, the complete source code of the modified version must be made available.

rvmn, 2023-03-19:
It does not say where and how, if you spent a minute of time clicking or even googling you can  see the source is for quite a time available, i just dont link to it yet as it is under a lot of work and i have no time yet setting up and finishing the site and docs, and those are for sure required. it is still a project. 
for completeness i will in the next days also release the superslicer version code, that should be released also i agree.

derlaft, 2023-03-19:
>  if you spent a minute of time clicking or even googling you can see the source is for quite a time available

Which source code? You are writing directly in the README that the code is not currently available. Not even sure what I am supposed to google for.

> i just dont link to it yet as it is under a lot of work and i have no time yet setting up and finishing the site and docs, and those are for sure required. it is still a project.

That's not really important, you are obliged to publish the source code of a derivative work. 

> It does not say where and how

Except it does:

     d) Convey the object code by offering access from a designated
    place (gratis or for a charge), and offer equivalent access to the
    Corresponding Source in the same way through the same place at no
    further charge.  You need not require recipients to copy the

rvmn, 2023-03-19:
The repo has been updated with links.

closed 2023-03-19 by rvmn

derlaft, 2023-03-19:
Awesome, thanks a lot