rvmn/SuperPleccer #11: Bridging perimeters should Always print before bridging infill

nhopfen, 2023-04-06:
This is related to the arc infill not anchoring to anything and drooping down on parts on 4 legs such as this https://thangs.com/designer/models/3d-model/62763 [...]

**Describe the solution you'd like**
On bridges, perimeters should ALWAYS be printed before infill, to provide an anchor point. This should be FORCED regardless of the "infill before perimeters" or the "external perimeters first" settings.

At the moment infill always prints before, whatever your settings, which causes prints to fail.

Keep up the great work

nhopfen, 2023-04-06:
[image]
[image]
[image]
[image]

nhopfen, 2023-04-06:
on a free-floating part, this outer bridging perimeter would then droop down but in my experience, this just means the loop could possibly act as a makeshift support layer, while also creating air turbulence that potentially throws the cooling back up towards the infill while it prints. and then as long as the infill is nice subsequent layers correct themselves quickly. Some people might prefer an toggleable option named "Bridging perimeters first"

nhopfen, 2023-04-09:
Sorry, my fault, I missed the "Overhang infill before perimeter" check box

closed 2023-04-09 by nhopfen

rvmn, 2023-04-18:
ok, i missed this issue, sorry, i thought github would inform me on issues automatically..

yeah, when perimeters should be printed is a bit of a subjective issue, but i think overhanging parts are always best infill first, also bridges, because they should not rely on the perimeters as anchors but on correct support lines/pillars below it. 
the perimeters/walls should just be to give a nice wall, and that works best if printed last (so anchored well on infill).

rvmn, 2023-04-19:
New feature in the coming version is autogenerated bridge supports (infill first):[image]