rvmn/SuperPleccer #12: Correct purge temperature when printing with different temperature materials

amatulic, 2023-04-08:
### What happened?

When current filament is being retracted and cool down is being performed, the next filaments temperature is set and causing the current filament to get uncontrollable strings that causes issues when the filament is being loaded again.
New filament temperature should not be set before current filament is gone from the printer head.
New filament must not be allowed at the printer head until its correct temperature is achieved.

### Project file & How to reproduce

Just make a project with filaments with different temperature settings, for example a PLA part with PETG supports.

Observe that when switching from PETG (high temperature) to PLA (lower temperature) the low temperature is set before the PETG is purged from the nozzle, causing the nozzle to jam because it is too cool to purge the PETG.

h2. I have fixed this issue.

The fix is in a PrusaSlicer pull request here: https://github.com/prusa3d/PrusaSlicer/pull/9737

I thought it would be helpful to users of this fork, which I believe is intended for multi-material printing.

### Operating system

Any

### Printer model

Prusa MK3S + MMU2S

rvmn, 2023-04-18:
that is smart,  is going in the next version for sure! (due in about two weeks, end of april). 
thanks a bunch!!!

amatulic, 2023-04-18:
You're welcome, and I hope this gets into PrusaSlicer soon too.

I just edited the issue description for better clarity. I can't believe I wrote that rambling mess earlier.