rvmn/SuperPleccer #14: Can't compile RC, neither can Actions

TekuSP, 2023-05-17:
Hey,

I am trying to compile RC version and I end up with
`10>LINK : fatal error LNK1104: cannot open file 'libexpat.lib'`

and

`7>C:\Users\richa\source\repos\SuperPleccer\src\libslic3r\SupportMaterial.cpp(1567,19): error C2666: 'Slic3r::ConfigOptionEnum<Slic3r::InfillPattern>::operator ==': 2 overloads have similar conversions`

I am using Visual Studio 2022 Preview, but I tried it on 2019 with no luck as well.
What I checked, Actions are not successful either.
Domain for PLEC community is not working, so I am left in void.

Is there any way I can gain binary or get into Discord, or anything to get this slicer?

TekuSP, 2023-05-17:
Okay, found hidden Releases, I have binary. But still, compile errors.

rvmn, 2023-05-19:
@TekuSP  
I had to retry a lot of times and never succeeded a built on Windows 11, only on Win10, but I was smart enough to make a backup copy of the build_deps: [link](https://rvmn.stackstorage.com/s/y6cPOd1WoiM5Y0jt)

pleccer, 2023-05-19:
@TekuSP  
I had to retry a lot of times and never succeeded a built on Windows 11, only on Win10, but I was smart enough to make a backup copy of the build_deps: [link](https://rvmn.stackstorage.com/s/y6cPOd1WoiM5Y0jt)

Mysli0210, 2023-05-31:
I also tried to compile both master and Alpha5_prerelease on win10 with vs 2019 community
However i dont get the cannot open libexpat.lib error. But rather just the "InfillPattern>::operator ==': 2" error.

I even tried with pleccers build_deps but to no avail.

ristiisa, 2023-06-15:
idk I got is compiling on windows with some minor tweaks'n'hacks https://github.com/rvmn/SuperPleccer/pull/17

rvmn, 2023-09-28:
> idk I got master compiling on windows with some minor tweaks'n'hacks #17

@ristiisa i tried with the merged version but i still run into expat linking problems (VS2019 on win11). could i ask you to make a zip of the build you made? or did you change stuff in the code and dont want to share? anyhow, if you'd be so kind to share a build, id be very thankful!

ristiisa, 2023-10-05:
> @ristiisa i tried with the merged version but i still run into expat linking problems (VS2019 on win11). could i ask you to make a zip of the build you made? or did you change stuff in the code and [...]

I made no changes beside the PR, I just wanted to see some features in action. So... I built from the command line just now like this:
```
> build_win.bat -s app -d=..\SuperPleccer-deps
```

And here is the end result from `.\SuperPleccer\build\src\RelWithDebInfo`
[release-without-dbg.zip](https://github.com/rvmn/SuperPleccer/files/12818039/release-without-dbg.zip)

rvmn, 2023-10-05:
@ristiisa Awesome, thanks that did the job! tested and uploaded the result to releases. 
Closing this issue!