rvmn/SuperPleccer #15: bridges run perpendicular to the desired direction

SurfBlaster, 2023-05-30:
All bridges run perpendicular to the desired direction - that is, along the long edges of the pit.

Maybe this can be fixed with some slicer settings, but I did not find such settings.

https://i.ibb.co/T0qZhTR/bridge-sp-wtf.png

ver. alpha5_prerelease

pleccer, 2023-05-30:
You can change the general setting of all the bridges using the bridge angle setting under infill. 
in the current dev state i set it so it will choose the direction that causes shortest bridging, i will push it in a few days

SurfBlaster, 2023-06-07:
Thanks, this really helped as a temporary solution!

infill - advanced - angle - bridging = 45

Offtopic:

However, printing my part is still not possible. Most arc-supports start in the air. And with a change in this angle, there are even more of them. And besides, an unexpected effect, changing this angle has caused the center of the support circles to now shift from the center of the polygon to its corner.

before/after

[image]
[image]

SurfBlaster, 2023-07-09:
> i will push it in a few days

My apologies, is there any progress?

rvmn, 2023-07-10:
yes, it is taking a bit longer than i was planning, apologies for that. 
i am spending the last hand on this tomorrow and the next days, ETA is before the end of the week.
thanks for the bump

j-snijder, 2023-07-15:
Hi, any idea on what timetable for the new windows compiled version ?Doing a great jib, thanks for all your effort SP beats PS by miles 😁