rvmn/SuperPleccer #16: Ready compiled Windows version ?

j-snijder, 2023-06-01:
I know its not supposed to be here but....

Is there any ready compiled windows version (alpha or Beta) we can download to test ?

I am espcially interested in the arc overhang mode and the color picking mode.

Why ? Because I can't find a slicer that is able to handle the 4in1 out hotends whgich are ment to use YMC-W as it should and SuperPleccer seems to be able to do that.

Kind regards, and please help :)

SurfBlaster, 2023-06-06:
Why don't you run the Linux version under virtualbox just yet?

j-snijder, 2023-06-12:
Hi :)

For me to setup linux under virtual box is a bit more then I can chew, sorry,too old for that.

I like pleccer, last beta I have found here on github somewhere and installed it, its way more and way better the Prusaslicer and I like to go on with pleccer, but as I can't find a more recent beta (or rc) I ask this here.

I know its a lot of work to each time compile a windows 64 bit version with each beta (or better still a nightly)  but as I am activ on some fb pages regarding the Zonestar Z9V5 I'd like to get some of these users to test PLeccer too and perhaps we can achieve better results then with Prusaslicer, as I expect it will.

Hope you, or anyone else will help out and do the builds for us :)

kind regards, Jan Snijder

Sveeeeeen, 2023-06-17:
Also waiting for the Windows version :)

Mysli0210, 2023-07-05:
I'm also waiting for it.
I tried to run the linux version under wsl but i couldn't get it running, but then again i have no experience with wsl, but do use debian on other machines

j-snijder, 2023-07-27:
Bumping this :)

Please pre-compile a Wndows version, we need it :)

rvmn, 2023-08-03:
Hi to you too, good day! Always good to see a fellow dutchman ;-)

I have been in a bit of limbo as to whether to spend more time on getting the windows build working, or fixing
the multi-color printing code and making that work. That last bit is what you also could use, and that is programmed in the PrusaSlicer fork called ['Pleccer'](https://github.com/pleccer/Pleccer). That fork has a working Github workflow and produces windows build fine so I dont have to spend time on the building stuff there :D

Tell me what you would like to see, apart from a basic gcode producing slicer with color-printing (in any color system).
As I started from scratch after a failed merge of the Prusa 2.6 into the old codebase I can see if it is easy enough to include. I will put Arc overhang and bridging imrovements in it again at the least.
Regarding time, it might be a week but it is at max two weeks, so 14 august at the last. Thats for the new Pleccer version.

Another thing: I have made a discord for Pleccer/SuperPleccer, which is still empty as of now, but feel free to put your ideas and related stuff there instead of in this issue, it makes sense to keep purely technical stuff in github.
Feel free to join at [discord](https://discord.gg/dERwmuF4)

j-snijder, 2023-08-04:
Hi, good day to you too.

I just checked on pleccer but there is no ready built windows version to find there.

As I am not much of a coder, nor builder, I am not familiar with how to compile a windows version myself, thats why I asked (and keep on asking) for a prebuilt windows version.

Is there anyway you, or anyone else you know, can build that for us as regular 3D printing fanatics ?

I hope so 😊

Kind regards, Jan

Verzonden vanuit Mail voor Windows

Van: rvmn
Verzonden: donderdag 3 augustus 2023 21:32
Aan: rvmn/SuperPleccer
CC: j-snijder; Author
Onderwerp: Re: [rvmn/SuperPleccer] Ready compiled Windows version ? (Issue #16)

Hi to you too, good day! Always good to see a fellow dutchman ;-)
I have been in a bit of limbo as to whether to spend more time on getting the windows build working, or fixing
the multi-color printing code and making that work. That last bit is what you also could use, and that is programmed in the PrusaSlicer fork called 'Pleccer'. That fork has a working Github workflow and produces windows build fine so I dont have to spend time on the building stuff there :D
Tell me what you would like to see, apart from a basic gcode producing slicer with color-printing (in any color system).
As I started from scratch after a failed merge of the Prusa 2.6 into the old codebase I can see if it is easy enough to include. I will put Arc overhang and bridging imrovements in it again at the least.
Regarding time, it might be a week but it is at max two weeks, so 14 august at the last. Thats for the new Pleccer version.
Another thing: I have made a discord for Pleccer/SuperPleccer, which is still empty as of now, but feel free to put your ideas and related stuff there instead of in this issue, it makes sense to keep purely technical stuff in github.
Feel free to join at discord
—

Dit e-mailbericht is met Avast-antivirussoftware gecontroleerd op virussen.
www.avast.com

j-snijder, 2023-08-04:
BTW, FYI.

The pleccer community site is marked as dangerous and will not load as its protected by virusscanner.

Any idea why I get a URL scam warning on that ?

Greets, Jan

Verzonden vanuit Mail voor Windows

Van: rvmn
Verzonden: donderdag 3 augustus 2023 21:32
Aan: rvmn/SuperPleccer
CC: j-snijder; Author
Onderwerp: Re: [rvmn/SuperPleccer] Ready compiled Windows version ? (Issue #16)

Hi to you too, good day! Always good to see a fellow dutchman ;-)
I have been in a bit of limbo as to whether to spend more time on getting the windows build working, or fixing
the multi-color printing code and making that work. That last bit is what you also could use, and that is programmed in the PrusaSlicer fork called 'Pleccer'. That fork has a working Github workflow and produces windows build fine so I dont have to spend time on the building stuff there :D
Tell me what you would like to see, apart from a basic gcode producing slicer with color-printing (in any color system).
As I started from scratch after a failed merge of the Prusa 2.6 into the old codebase I can see if it is easy enough to include. I will put Arc overhang and bridging imrovements in it again at the least.
Regarding time, it might be a week but it is at max two weeks, so 14 august at the last. Thats for the new Pleccer version.
Another thing: I have made a discord for Pleccer/SuperPleccer, which is still empty as of now, but feel free to put your ideas and related stuff there instead of in this issue, it makes sense to keep purely technical stuff in github.
Feel free to join at discord
—

Dit e-mailbericht is met Avast-antivirussoftware gecontroleerd op virussen.
www.avast.com

Mysli0210, 2023-08-06:
@rvmn 
For me at least the thing that draws me to superslicer is that i cannot tune prusa slicer enough for dimensional accuracy 
(there are just missing essential features from my point of view, how people get good prints with volcano nozzles from prusaslicer is beyond me).

Then what drew me to superpleccer is the arc overhangs (which i'd love to be able to tune, cause they surely work but could be better), not multicolor printing as i dont have any mmu or other multi filament devices.

as for the pleccer build under actions, it seems that the latest build simply just expired hence being unavailable.

rvmn, 2023-09-23:
Please check the [crowdfund campaign](https://github.com/rvmn/SuperPleccer/discussions/27)
Thank you!