rvmn/SuperPleccer #19: How to open SuperPleccer-macOS

teefixx, 2023-07-26:
Hi all

I tried to update from alpha4 to the latest release but could not get how to use the file "SuperPleccer-macOS". Renaming it to dmg does not help.

Any hints would be helpful.

Thank you

rvmn, 2023-07-26:
Hi @teefixx
You can rename the file to. zip and unzip it.
It is just alpha5, I'm still working on the new version.
Good printing,

rvmn

j-snijder, 2023-07-27:
Hi,Do you also build a windows version ? Please. ?Regards, J. Snijder

FloatyPrints, 2023-08-26:
Anyone figured out how to use it now on Mac? 
Actually can't figure out how to use it

GamerMechanic1981, 2023-09-02:
Cant open it either.  Running 10.14.6 on a 2010 mac pro.  It doesnt see it as a valid executable application file

rvmn, 2023-09-03:
Sorry, it was unclearly explained. You need to download with non-safari or turn off automatic unzipping of downloads, and rename the file ending by removing the .zip ending. And confirm changing the type.