rvmn/SuperPleccer #25: Please provide an executable Windows 64-bit version of the program

laptop-laptop, 2023-09-20:
**Is your feature request related to a problem? Please describe.**

Hello, I cannot find it anywhere. Could you please post a pre-compiled, executable Windows 10 64-bit version? Thanks

rvmn, 2023-09-23:
https://github.com/rvmn/SuperPleccer/discussions/27

SurfBlaster, 2023-10-12:
Thank you for providing the windows version (2.5.59.2)! One annoying error immediately appeared: file «SuperPleccer_192px.png» is missing and therefore the program crashes when it wants to show an error. Just put this file in «"c:\Program Files\SuperPleccer\resources\icons\» and everything will be a little better.
[image]
[SuperPleccer_192px_png.zip](https://github.com/rvmn/SuperPleccer/files/12886204/SuperPleccer_192px_png.zip)

rvmn, 2023-11-04:
@SurfBlaster Thanks for your fix. It must be some download or unzip problem, in the zip file it seems to already have the file in the resources/icons folder. Or did you built the slicer from source?