rvmn/SuperPleccer #26: Is  the SuperPleccer project terminated?

laptop-laptop, 2023-09-22:
### What happened?

I see several people requested for the Windows executable program for the past few months but nothing has been done. Is this project terminated?

### Project file & How to reproduce

none

### Version

2.4

### Operating system

window 10

### Printer model

Prusa i3MK3S+

rvmn, 2023-09-23:
OK, good you ask, because things are a bit complicated.
Even though I would like to get it to work there are two complications that prevent it from happening:
A) the compiling for mac and windows always leads to dependency problems and this is hard to solve because it is using an older dependency chain than say Prusa/Bambu slicer and as such cannot use their code to compile the app.
B) I personally am not finding the time to solidly fix this, and as such only have time to try 'small' in-place fixes, which should work, theoretically, but are not, and don't need the whole depency chain to be updated.  
Bottom-line: I would like it to work but I cannot.
The only solution in my opinion would be to either:
A) open a bounty on fixing the building and pay someone to jump in and fix it, or:
B) start a crowdfund and see if there are enough people wanting all the proposed features (multi-color, no-support, purgeless printing) and spend it to hire someone(s) to implement them. 
I wish it was simpler, but it simply isn't.

rvmn, 2023-09-23:
I thought a bit about it and did a little research.
My finding is that it will require about $1000 to get the main features worked out using an outsourcing approach (hiring someone from a platform like fiverr). 
That would mean 100 people would need to put $10 in the project.
I'll this weekend make a poll to find if enough people can chip that in, link will be added.

rvmn, 2023-09-23:
Please fill in the [poll](https://github.com/rvmn/SuperPleccer/discussions/27)
Thank you!

elementalbeast, 2023-09-27:
it runs on windows for me

elementalbeast, 2023-09-27:
can you point me to the file in the source code that controls the arc overhangs implementation

elementalbeast, 2023-09-27:
i didn't do anything to the file all i did was download it.  was pleccer not running with arc overhangs for you?

rvmn, 2023-09-28:
> @elementalbeast Are you saying you have a working Windows copy of Pleccer with arc overhangs then ? Would you, or rvmn, share this cooy with us ? That would be awesome 😁

@j-snijder Pleccer is the PrusaSlicer-based version, that works fine on Windows, it can be downloaded on the [releases page](https://github.com/rvmn/SuperPleccer/releases). version alpha4-2 being the latest. it adds arcs by default and doesnt have any settings on them, so just use and you get arcs, and dont forget to set the bridge speed low, it doesnt have arc speed seperately (yet) i think.