rvmn/SuperPleccer #31: Any intentions on updating the project?

Wang-Thunder, 2024-02-16:
First of all, I would like to say great work and thanks for such an awesome contribution to the community :) I have tried probably a dozen different slicers and I just keep coming back to SuperPleccer because its still one of those feature rich and easy to use slicers around.

I'm curious if there are any intentions on continuing the project and updating it to include other modern features from PrusaSlicer, etc. Obviously maintaining a project like this takes considerable time and its totally understandable if the time just isn't available. Just trying to determine if I should transition to something else or stick with it :) 

Thanks again!

rvmn, 2024-02-17:
Hi, well yes and no, atm it is a lot of work to tranfer features from Prusa to Super because it is based on a much older version and missing a lot of changes needed. But, supermerill is atm working on moving superslicer to 2.7 and then i will update superpleccer to also that version.
In the meantime i am working on Iris, a multicolor fork of Prusa for mixing color printers. See [Iris](https://github.com/plec3d/Iris)

rvmn, 2024-02-17:
[Release alpha version](https://github.com/plec3d/Iris/releases/tag/2.7_alpha1_prerelease)